natario1/CameraView #1214: Internal Camera1 error. 1

IvanovIgorBelarus, 2023-01-18:
- CameraView version: 2.7.1
- Camera engine used: camera2
- Xiaomi Redmi Note11 pro, API 31

condition:
 if (cameraType == CameraType.BACK_CAMERA) {
            cameraView?.facing = Facing.BACK // get error
        } else {
            cameraView?.facing = Facing.FRONT   //currently working
        }

### XML layout
 <com.otaliastudios.cameraview.CameraView
            android:id="@+id/camera"
            android:keepScreenOn="true"
            android:layout_width="match_parent"
            android:layout_height="match_parent"
            android:layout_gravity="center"
            app:cameraUseDeviceOrientation="false"
            app:cameraEngine="camera2"
            app:cameraMode="picture"
            app:cameraPreview="surface"
            app:cameraAudio="off" />

### Logs
E/Camera: Error 1
E/CameraEngine: Internal Camera1 error. 1
E/AndroidRuntime: FATAL EXCEPTION: CameraViewEngine
    Process: com.alseda.vtbbank, PID: 21644
    com.otaliastudios.cameraview.CameraException: java.lang.RuntimeException: Internal Camera1 error. 1
        at com.otaliastudios.cameraview.engine.Camera1Engine.onError(Camera1Engine.java:92)
        at android.hardware.Camera$EventHandler.handleMessage(Camera.java:1369)
        at android.os.Handler.dispatchMessage(Handler.java:106)
        at android.os.Looper.loop(Looper.java:237)
        at android.os.HandlerThread.run(HandlerThread.java:67)
     Caused by: java.lang.RuntimeException: Internal Camera1 error. 1
        at com.otaliastudios.cameraview.engine.Camera1Engine.onError(Camera1Engine.java:82)
        at android.hardware.Camera$EventHandler.handleMessage(Camera.java:1369) 
        at android.os.Handler.dispatchMessage(Handler.java:106) 
        at android.os.Looper.loop(Looper.java:237) 
        at android.os.HandlerThread.run(HandlerThread.java:67)

trinadhthatakula, 2023-03-01:
We are facing the same issue in android 11 moto devices

trinadhthatakula, 2023-03-01:
I have tried to change the engine to camera2 and enabled experimental camera and this was the log

10:49:08.728  W  CameraDevice 0 died unexpectedly
10:49:08.729  E  CameraDevice.StateCallback reported an error: 5
10:49:08.730  E  FATAL EXCEPTION: CameraViewEngine
                 Process: com.rstech.camerascanner.pdfscanner.camscanner.easy.scan.camscanner, PID: 4296
                 com.otaliastudios.cameraview.CameraException
                 	at com.otaliastudios.cameraview.engine.Camera2Engine$2.onError(Camera2Engine.java:463)
                 	at android.hardware.camera2.impl.CameraDeviceImpl$14.run(CameraDeviceImpl.java:2263)
                 	at android.os.Handler.handleCallback(Handler.java:938)
                 	at android.os.Handler.dispatchMessage(Handler.java:99)
                 	at android.os.Looper.loop(Looper.java:223)
                 	at android.os.HandlerThread.run(HandlerThread.java:67)
10:49:08.731  E  EXCEPTION: Handler thread is gone. Replacing.
10:49:08.733  E  EXCEPTION: Scheduling on the crash handler...
10:49:08.759  E  EXCEPTION: Got CameraException. Since it is unrecoverable, executing destroy(false).
10:49:14.761  E  DESTROY: Could not destroy synchronously after 6 seconds. Current thread: Thread[main,5,main] Handler thread: Thread[CameraViewEngine,5,main][4424]
10:49:14.762  E  DESTROY: Trying again on thread: Thread[CameraViewEngine,5,main][4436]
10:49:20.763  E  DESTROY: Could not destroy synchronously after 6 seconds. Current thread: Thread[main,5,main] Handler thread: Thread[CameraViewEngine,5,main][4436]
10:49:20.763  E  EXCEPTION: Got CameraException. Dispatching to callback.
10:49:20.766  I  Skipped 1081 frames!  The application may be doing too much work on its main thread.

we have only observed this issue in android 11 devices

klower, 2023-08-14:
same issue

duyhungtnn, 2023-08-26:
I am facing the same issues on Camera1Engine 
Xiaomi Redmi Note7 pro, API 33