natario1/CameraView #1224: How Flip camera horizontally

Child0fTheSun, 2023-02-23:
I use snapshots to take photos and videos and would like to be able to flip the camera horizontally. I tried to do it with filters, but without success. Anyone could give me a hint?

Im using version 2.7.2

Child0fTheSun, 2023-02-23:
Use custom filter

`public class CameraFlipEffect extends BaseFilter {
    @Override
    protected void onPreDraw(long timestampUs, @NonNull float[] transformMatrix) {
        android.opengl.Matrix.rotateM(transformMatrix, 0, 180, 0, 1, 0);
        android.opengl.Matrix.translateM(transformMatrix, 0, -1, 0, -1);
        super.onPreDraw(timestampUs, transformMatrix);
    }

    @NonNull
    @Override
    public String getFragmentShader() {
        return createDefaultFragmentShader();
    }
}`

closed 2023-02-23 by Child0fTheSun

jaepeng, 2024-05-16:
GoooD!