natario1/CameraView #1225: How to get mime type and extension of taken videos and pictures?

waterdrake, 2023-02-26:
Is there a way to get the the mime type and extension of a taken video? I can get the file but it does not seem to have an extension or mime type:

```
            @Override
            public void onVideoTaken(@NonNull VideoResult result) {
                File file = result.getFile();
            }
```

Also, on that note, is there a way to get the mime type and extension of a taken picture?
```

            @Override
            public void onPictureTaken(PictureResult result) {
            }
```

EzequielAdrianM, 2023-09-11:
When you configure the camera settings, you can tell the Camera API what resolution, format, and compression to use, so you should already know it.

waterdrake, 2023-11-05:
@EzequielAdrianM Can you point me to where I can learn about how to configure those camera settings you mentioned?