natario1/CameraView #1231: cameraView.setFilter but screen is black

zhaichuankai, 2023-03-23:
### Describe the bug
Please add a clear description of what the bug is, **and** fill the list below.
- CameraView version: *2.6.4*
- Camera engine used: *camera1*
- Reproducible in official demo app: *yes*
- Device / Android version: *Pixel, API 28*
- I have read the [FAQ page](https://natario1.github.io/CameraView/about/faq): *yes*

### Expected behavior
when i add a custom filter,but the screen is black

### XML layout
Part of the XML layout with the CameraView declaration, so we can read its attributes.

```xml
        <com.otaliastudios.cameraview.CameraView
            android:id="@+id/camera"
            android:layout_width="match_parent"
            android:layout_height="0dp"
            android:layout_gravity="center"
            android:keepScreenOn="true"
            app:cameraAudio="off"
            app:cameraAutoFocusMarker="@string/cameraview_default_autofocus_marker"
            app:cameraEngine="camera1"
            app:cameraExperimental="true"
            app:cameraFacing="back"
            app:cameraFlash="off"
            app:cameraGestureLongTap="none"
            app:cameraGesturePinch="zoom"
            app:cameraGestureScrollHorizontal="filterControl1"
            app:cameraGestureScrollVertical="exposureCorrection"
            app:cameraGestureTap="autoFocus"
            app:cameraGrid="off"
            app:cameraMode="picture"
            app:cameraPlaySounds="true"
            app:cameraPreview="glSurface"
            app:layout_constraintBottom_toBottomOf="parent"
            app:layout_constraintDimensionRatio="1:1.33"
            app:layout_constraintTop_toTopOf="parent">

        </com.otaliastudios.cameraview.CameraView>
```
my filter:

import android.opengl.GLES20;

import androidx.annotation.NonNull;

import com.otaliastudios.cameraview.filter.BaseFilter;

import java.nio.FloatBuffer;

public class GPUImageBeautyFilter extends BaseFilter {
    public static final String BILATERAL_FRAGMENT_SHADER = "" +
            "   varying highp vec2 vTextureCoord;\n" +
            "\n" +
            "    uniform sampler2D inputImageTexture;\n" +
            "\n" +
            "    uniform highp vec2 singleStepOffset;\n" +
            "    uniform highp vec4 params;\n" +
            "    uniform highp float brightness;\n" +
            "\n" +
            "    const highp vec3 W = vec3(0.299, 0.587, 0.114);\n" +
            "    const highp mat3 saturateMatrix = mat3(\n" +
            "        1.1102, -0.0598, -0.061,\n" +
            "        -0.0774, 1.0826, -0.1186,\n" +
            "        -0.0228, -0.0228, 1.1772);\n" +
            "    highp vec2 blurCoordinates[24];\n" +
            "\n" +
            "    highp float hardLight(highp float color) {\n" +
[... 119 lines cut ...]
        brightnessLocation = GLES20.glGetUniformLocation(programHandle, "brightness");
        singleStepOffsetLocation = GLES20.glGetUniformLocation(programHandle, "singleStepOffset");
    }

    @Override
    public void onDestroy() {
        super.onDestroy();
        paramsLocation = -1;
        brightnessLocation = -1;
        singleStepOffsetLocation = -1;
    }

    @Override
    protected void onPreDraw(long timestampUs, @NonNull float[] transformMatrix) {
        super.onPreDraw(timestampUs, transformMatrix);
        float[] vector = new float[4];
        vector[0] = 1.0f - 0.6f * beautyLevel;
        vector[1] = 1.0f - 0.3f * beautyLevel;
        vector[2] = 0.1f + 0.3f * toneLevel;
        vector[3] = 0.1f + 0.3f * toneLevel;
        GLES20.glUniform2fv(singleStepOffsetLocation, 1, FloatBuffer.wrap(new float[] {2.0f / width, 2.0f / height}));
        GLES20.glUniform4fv(paramsLocation, 1, FloatBuffer.wrap(vector));
        GLES20.glUniform1f(brightnessLocation, 0.6f * (-0.5f + brightLevel));
    }
}

zhaichuankai, 2023-03-23:
yeah  i sloved it

zhaichuankai, 2023-03-23:
For advanced users, child filters need to read from GLES20.GL_TEXTURE_2D instead of the typical GLES11Ext.GL_TEXTURE_EXTERNAL_OES. To achieve this, we get your fragment shader String and replace any "samplerExternalOES" with "sampler2D". This is a hack that might cause issues with specific shader implementations.

zhaichuankai, 2023-03-23:
replace  sampler2D to samplerExternalOES    and add "#extension GL_OES_EGL_image_external : require\n"

closed 2023-03-23 by zhaichuankai

353338110, 2023-04-13:
> replace sampler2D to samplerExternalOES and add "#extension GL_OES_EGL_image_external : require\n"

I also do this,but it still balck.can you give complete code