natario1/CameraView #1236: NullPointerException at GlTextureDrawer.setFilter

EzequielAdrianM, 2023-05-15:
```
Fatal Exception: java.lang.NullPointerException: Attempt to write to field 'e9.b i9.e.d' on a null object reference
       at com.otaliastudios.cameraview.internal.GlTextureDrawer.setFilter(GlTextureDrawer.java:49)
       at com.otaliastudios.cameraview.picture.SnapshotGlPictureRecorder.onRendererFilterChanged(SnapshotGlPictureRecorder.java:122)
       at com.otaliastudios.cameraview.picture.SnapshotGlPictureRecorder$1.onRendererFilterChanged(SnapshotGlPictureRecorder.java:90)
       at com.otaliastudios.cameraview.preview.GlCameraPreview$2.run(GlCameraPreview.java:303)
       at android.opengl.GLSurfaceView$GLThread.guardedRun(GLSurfaceView.java:1500)
       at android.opengl.GLSurfaceView$GLThread.run(GLSurfaceView.java:1270)
```

It looks like somehow, at line (49) of GlTextureDrawer.java, the function setFilter is receiving a null argument thus causing NonNull annotation to raise the exception.

```
public void setFilter(@NonNull Filter filter) {
        mPendingFilter = filter;
    }
```

Maybe the NonNull annotation should be removed? Can the app continue working if mPendingFilter is null?
This exception appeared at random just in one device in production environment.

ShakeelAziz, 2023-07-26:
Facing the same issue, Any one know how to solve this?

big-guogai, 2023-12-28:
Facing the same issue, Any one know how to solve this?

abbas427, 2024-05-21:
I'm also facing same issue.