natario1/CameraView #1256: Fatal Exception: android.media.MediaCodec$CodecException: start failed

faldujaldeep, 2023-10-27:
### Describe the bug
Please add a clear description of what the bug is, **and** fill the list below.
- CameraView version: [*version number*](com.otaliastudios:cameraview:2.6.4)
- Device / Android version: Redmi Note 12 5G, Redmi Note 12, 
Poco X5 5G

<com.otaliastudios.cameraview.CameraView
            android:id="@+id/cameraView"
            android:layout_width="match_parent"
            android:layout_height="match_parent"
            android:layout_gravity="center"
            android:keepScreenOn="true"
            app:cameraAudio="off"
            app:cameraAutoFocusMarker="@string/cameraview_default_autofocus_marker"
            app:cameraEngine="camera2"
            app:cameraExperimental="true"
            app:cameraFacing="front"
            app:cameraFlash="auto"
            app:cameraGestureLongTap="none"
            app:cameraGesturePinch="zoom"
            app:cameraGestureScrollHorizontal="filterControl1"
            app:cameraGestureScrollVertical="exposureCorrection"
            app:cameraGestureTap="autoFocus"
            app:cameraGrid="off"
            app:cameraMode="video"
            app:cameraPlaySounds="true"
            app:cameraPreview="glSurface"
            app:cameraVideoMaxSize="3670016"
            app:cameraWhiteBalance="auto" />

Fatal Exception: android.media.MediaCodec$CodecException: start failed
       at android.media.MediaCodec.native_start(MediaCodec.java)
       at android.media.MediaCodec.start(MediaCodec.java:2322)
       at com.otaliastudios.cameraview.video.encoding.VideoMediaEncoder.onPrepare(VideoMediaEncoder.java:84)
       at com.otaliastudios.cameraview.video.encoding.TextureMediaEncoder.onPrepare(TextureMediaEncoder.java:98)
       at com.otaliastudios.cameraview.video.encoding.MediaEncoder$1.run(MediaEncoder.java:191)
       at android.os.Handler.handleCallback(Handler.java:942)
       at android.os.Handler.dispatchMessage(Handler.java:99)
       at android.os.Looper.loopOnce(Looper.java:211)
       at android.os.Looper.loop(Looper.java:300)
       at android.os.HandlerThread.run(HandlerThread.java:67)

AlyMBarakat, 2024-07-04:
Having the same crash

doanvu2000, 2024-07-29:
are you resolve exception? I have same crash

khiemdnt, 2024-08-28:
> are you resolve exception? I have same crash
have you fixed it. how to fix?

doanvu2000, 2024-10-11:
@khiemdnt I suround function onPrepare of VideoMediaEncoder.java with try/catch
[image]