natario1/CameraView #1267: Flash Bug

Shahidzbi4213, 2024-01-07:
**Flash Auto** and **Flash On** is not working with camera2.

Device **Redmi Note 12** 
Android Version **14**

EzequielAdrianM, 2024-01-08:
Hi, if you use the Picture Snapshot and Video Snapshot, it is taking a screenshot directly from the OpenGL surface. This means that photo and video are taken much faster, but the Android camera was never actually ordered to take anything, thats why you see no flash. If you want all the camera functionalities you must use the takePicture and recordVideo (no snapshot) versions.

Shahidzbi4213, 2024-01-09:
Thanks buddy for clearing that up ☺️

EzequielAdrianM, 2024-01-09:
According to the [documentation](https://natario1.github.io/CameraView/docs/watermarks-and-overlays) watermark only works when using snapshot. Anyway, you can manually attach your watermark(bitmap) overlay into the picture using a canvas. And for watermark in video I highly recommend using [LiTr](https://github.com/linkedin/LiTr). We use it in production for overlay bitmap, gif, and stickers in our videos and works like a champ.

Shahidzbi4213, 2024-01-09:
Actually my task is to show timer running on the video with geo-location tag is it possible to achieve using the recommended library?

closed 2024-01-13 by Shahidzbi4213