natario1/CameraView #1282: Not able to record video in android 14 in some devices

rakeshrajput537, 2024-07-29:
### Describe the bug
I have two android 14 devices one is oneplus and another one is vivp.
in my oneplus it is working fine but failing in vivo.

### To Reproduce
Start video recording in android 14 in vivo phone.

### Expected behavior
Should record video as expect 

### XML layout
Part of the XML layout with the CameraView declaration, so we can read its attributes.

    <androidx.constraintlayout.widget.ConstraintLayout
        android:layout_width="match_parent"
        android:layout_height="match_parent">

        <com.otaliastudios.cameraview.CameraView
            android:id="@+id/camera"
            android:layout_width="match_parent"
            android:layout_height="match_parent"
            android:keepScreenOn="true"
            app:layout_constraintBottom_toTopOf="@+id/bottomPanel"
            app:layout_constraintTop_toTopOf="parent" />
 </androidx.constraintlayout.widget.ConstraintLayout>

### Logs

onVideoResult result is null: something went wrong. java.lang.RuntimeException: start failed.
                                                                                                    java.lang.RuntimeException: start failed.
                                                                                                    	at android.media.MediaRecorder._start(Native Method)
                                                                                                    	at android.media.MediaRecorder.start(MediaRecorder.java:1376)
                                                                                                    	at com.otaliastudios.cameraview.video.FullVideoRecorder.onStart(FullVideoRecorder.java:314)
                                                                                                    	at com.otaliastudios.cameraview.video.VideoRecorder.start(VideoRecorder.java:80)
                                                                                                    	at com.otaliastudios.cameraview.engine.Camera1Engine.onTakeVideo(Camera1Engine.java:426)
                                                                                                    	at com.otaliastudios.cameraview.engine.CameraBaseEngine$5.run(CameraBaseEngine.java:605)
                                                                                                    	at com.otaliastudios.cameraview.engine.orchestrator.CameraStateOrchestrator$3.run(CameraStateOrchestrator.java:100)
                                                                                                    	at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$1.call(CameraOrchestrator.java:84)
                                                                                                    	at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$1.call(CameraOrchestrator.java:81)
                                                                                                    	at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$3.run(CameraOrchestrator.java:152)
                                                                                                    	at com.otaliastudios.cameraview.internal.WorkerHandler.run(WorkerHandler.java:137)
                                                                                                    	at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator.execute(CameraOrchestrator.java:147)
                                                                                                    	at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator.access$100(CameraOrchestrator.java:34)
                                                                                                    	at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$2.run(CameraOrchestrator.java:137)
                                                                                                    	at android.os.Handler.handleCallback(Handler.java:1013)
                                                                                                    	at android.os.Handler.dispatchMessage(Handler.java:101)
                                                                                                    	at android.os.Looper.loopOnce(Looper.java:226)
                                                                                                    	at android.os.Looper.loop(Looper.java:328)
                                                                                                    	at android.os.HandlerThread.run(HandlerThread.java:67)

Rohitrajkhorwal, 2024-07-29:
Are you getting this error with cameraEngine camera2 also?
And did you change its mode before recording the video?

`camera.mode = Mode.VIDEO`

rakeshrajput537, 2024-07-29:
Yes I change the mode to video,  I didn't try cameraEngine and camera2

rakeshrajput537, 2024-07-29:
How to use camera2 or camera engine?

AntonLogin0v, 2024-07-30:
I have the same problem

AntonLogin0v, 2024-08-08:
The problem is in the methods

mediaRecorder.setVideoSize(mProfile.videoFrameWidth, mProfile.videoFrame Height);
mediaRecorder.setVideoFrameRate(mProfile.videoFrameRate);

works for me, comment them out(download lib), but they affect the quality

source: https://blog.csdn.net/netwalk/article/details/17686993